Makerfabs/Makerfabs-ESP32-UWB #4: Antenna tuning / calibration accuracy

alastaira, 2022-01-19:
Hi,

The board design seems good, but I'm getting poor accuracy on the range readings (i.e. when the distance between anchor and tag is consistently reading around 50cm too high). From reading the Decawave documentation, there is an application note regarding antenna calibration, which suggests this could be resolved in either software or hardware design: https://www.decawave.com/wp-content/uploads/2018/10/APS014_Antennna-Delay-Calibration_V1.2.pdf - is that something you've accounted for at all, or can suggest how to incorporate in the library?

Cheers,

pbs-websuntangled, 2022-01-19:
Agreed.i get the same 

Also readings are somewhat inconsistent. Rotating the tag or anchor through 90 degrees can add or remove 200mm from the distance reported

Fbo06, 2022-11-03:
hello, got the same issue, do you have found a solution ? thank you for your response

alastaira, 2022-11-03:
> hello, got the same issue, do you have found a solution ? thank you for your response

No, I gave up I'm afraid. Unfortunately I think these boards are not useful for anything other than toys :(

franciscoQM, 2022-11-29:
I came across the same issue. Although I am constantly getting an error of about 0.6 - 0.7+ meters I tested in a distance of from 0-125 meters. I wonder if anyone has found a way of improving the error. For my application all I need is .5 meters