Makerfabs/Makerfabs-ESP32-UWB #8: Not possible to connect more than 5 anchors to 1 tag

esvelaren, 2022-10-10:
Whenever I try connecting more than 5 anchors to a tag the whole communication gets completely bugged and the ranging process between TAG and ANCHORS is basically stopped. I tried playing with the parameter MAX_DEVICES and LEN_DATA from DW1000Ranging.h but couldn't get it to work with more than 5 anchors at a time, which is a pitty. I really hope they take a look into this issue.

NoahM22, 2023-11-27:
@esvelaren Hey! We are running into the same issue at the moment. Were you ever able to make any development on this issue we are trying to do the same and ran into some hardware exceptions with over 4 anchors. Thanks!

robertocuervot, 2025-06-05:
**For anyone who encounter this problem:**
Refer to [this](https://github.com/thotro/arduino-dw1000/issues/341#issuecomment-2342902083) comment in the original library repository.