Makerfabs/Makerfabs-ESP32-UWB #9: Does this library work with the ESP32 - DW3000 ?

ThomasLengeling, 2022-11-12:
I noticed on the examples and documentation that it only mentions the DW1000 modules, does the library works as well with the ESP32-DW3000 version as well?
Many Thanks for the help

codingcatgirl, 2023-02-08:
Sure, the library for DWM3000 was added on Jun 29, 2022 and then promptly deleted in the next commit 3 days later. No idea what the heck the reason is for that, but if you simply look at the state after Jun 29 2022, you can find the library there. https://github.com/Makerfabs/Makerfabs-ESP32-UWB/tree/eac7783183e03086f484089bf349c6ac7480bd40

codingcatgirl, 2023-02-08:
It was moved to this repository: https://github.com/Makerfabs/Makerfabs-ESP32-UWB-DW3000