Makerfabs/Makerfabs-ESP32-UWB #14: The example code can't work with ESP32 UWB Pro with Display

willy20040711, 2023-03-28:
I try to use 3 ESP32 UWB Pro with Display boards. Two anchors and one for tag. 
The example code of [indoorIndoorPositioning](https://github.com/Makerfabs/Makerfabs-ESP32-UWB/tree/main/example/IndoorPositioning)  can't work.
The tag only return empty object like: {LINK:[]}

juanelog, 2023-10-26:
Hi Willy
Did you find any solution to this problem?

> I try to use 3 ESP32 UWB Pro with Display boards. Two anchors and one for tag. The example code of [indoorIndoorPositioning](https://github.com/Makerfabs/Makerfabs-ESP32-UWB/tree/main/example/IndoorPositioning) can't work. The tag only return empty object like: {LINK:[]}

I'm trying to do indoor positioning but I have the same result.
Do you have any recommendations?
Thank you

lewthomps, 2023-11-24:
same issue here! 

We're either of you able to find a solution to this problem?

Thanks!

lewthomps, 2023-11-24:
nevermind! switchting DW_CS pin to 21 fixed the problem!

scinfu, 2024-01-17:
not working for me change the pint to 21.
I have bought this: https://www.makerfabs.com/mauwb-dw3000-with-stm32-at-command.html