Makerfabs/Makerfabs-ESP32-UWB #16: Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1) using ESP32 anchor example code

lewthomps, 2023-11-24:
Hi All,

I'm using a Mac running Monterey and attempting to run the [example code](https://github.com/Makerfabs/Makerfabs-ESP32-UWB/blob/main/example/IndoorPositioning/anchor/anchor.ino) on Arduino IDE for the [ESP32 UWB](https://www.makerfabs.com/esp32-uwb-ultra-wideband.html), but it runs into this error:

`
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4008ac3a  PS      : 0x00060735  A0      : 0x80089bb2  A1      : 0x3ffbf18c  
A2      : 0x3ffb7d84  A3      : 0x3ffbd330  A4      : 0x00000004  A5      : 0x00060723  
A6      : 0x00060723  A7      : 0x00000001  A8      : 0x3ffbd330  A9      : 0x00000018  
A10     : 0x3ffbd330  A11     : 0x00000018  A12     : 0x3ffc2794  A13     : 0x00060723  
A14     : 0x007bf338  A15     : 0x003fffff  SAR     : 0x0000001e  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400841e1  LEND    : 0x400841e9  LCOUNT  : 0x00000027  
Core  1 was running in ISR context:
EPC1    : 0x400dfb6b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace: 0x4008ac37:0x3ffbf18c |<-CORRUPTED

Core  0 register dump:
`

I did not touch the code. It is straight copied and pasted from the git repository, which led me to believe it would work without a hitch!

It seems to have an issue with this line:
`
DW1000Ranging.initCommunication(PIN_RST, PIN_SS, PIN_IRQ);
`
But I can't figure out why or what needs to be changed to get this code to run correctly.

Any help would be greatly appreciated!!!

Best
Lewis

closed 2023-11-24 by lewthomps

kaiming-uw, 2024-07-09:
@lewthomps Hi Lewis, I encountered the same problem when running the ESP32 anchor code. I wonder if you have any insights on fixing this bug, thanks!

cicintrip, 2025-01-14:
how did you solve it? 
I'm having the same problem :(