Makerfabs/Makerfabs-ESP32-UWB #18: Issue with the limited uwb range in existing code

rachanaa04, 2024-08-05:
I am currently working on a project using the UWB modules from Makerfabs and have been utilizing the UWB codes provided in the official Makerfabs GitHub repository. According to the product specifications on your website, the expected range of these UWB modules is over 100 meters. However, in my tests, I am consistently getting a limited range of only 3.7 to 4 meters.
Here are some details of my setup:
1.I am using the same UWB codes as provided in the Makerfabs repository.
2.The environment for testing is open and free of obstacles that might interfere with the UWB signal.
3.I have followed all the setup instructions as per the documentation.

Any solutions for this ?

Vorstok, 2025-04-04:
Hi, I had a similar problem, you need to experiment with the driver DW1000.cpp

I got it working after replacing this:

byte DW1000Class::_preambleCode = PREAMBLE_CODE_16MHZ_4;
byte DW1000Class::_channel = CHANNEL_5;

It was replaced with this:

byte DW1000Class::_preambleCode = PREAMBLE_CODE_16MHZ_2;
byte DW1000Class::_channel = CHANNEL_1;

Juli-blip-creator, 2025-07-16:
Hi, I am having the same Problem. Also changed Channel and MHZ but it won´t work.