Makerfabs/Makerfabs-ESP32-UWB #20: Struggling to Achieve Promised Range with ESP32 UWB Pro High-Power Modules

Jo-drc, 2025-04-29:
I have purchased 5 ESP32 UWB Pro modules from Makerfabs, but unfortunately, I am not able to achieve the range stated on their website. I have followed all the instructions as described and even made changes in the DW1000.cpp file, replacing:

byte DW1000Class::_preambleCode = PREAMBLE_CODE_16MHZ_4;
byte DW1000Class::_channel = CHANNEL_5;

with:

byte DW1000Class::_preambleCode = PREAMBLE_CODE_16MHZ_2;
byte DW1000Class::_channel = CHANNEL_1;

But even after implementing it, I can only achieve a range of 8 meters. Moreover, the reported signal strength is around -70 dBm, even when the devices are as close as 20 cm to each other.

Thank you in advance for any advice or suggestions.

Juli-blip-creator, 2025-07-16:
Hi, I am having the exact same Problem if you found any solution please let me know! :)

Juli-blip-creator, 2025-07-18:
Hi, just found a solution!

Jo-drc, 2025-07-18:
Apologies for the delayed response. I wasn’t able to find a viable solution, so I’ve decided to put the project on hold indefinitely. May I ask what approach you’ve taken to resolve the issue?